nand2mario/nestang #2: Cannot Flash my Sipeed Primer 20k

Chandler-Kluser, 2022-10-24:
Since I have tried to flash my device with the new release v.0.4.0 my external flash memory is not working anymore.

I have tried to flash my second Sipeed Tang Primer 20k too, but the same thing happened. At that second time I recorded my screen to show you the output (you can download through [this link](https://drive.google.com/file/d/1_6bGbvuAw7Ae1G0UfD_5_8zEMoPUdtw7/view?usp=sharing))

I am not sure if the SPI flash chip has been bricked, yet.

I have tried to flash both with openFPGALoader and GoWin Official Programmer, none worked at all...

I have also compiled from source and flashed it, but it didn't work either...

The output I have got when I flashed the FPGA was:
```
openFPGALoader -b tangprimer20k -f nestang-0.4-lite.fs 
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Parse file Parse nestang-0.4-lite.fs: 
Done
DONE
Jtag frequency : requested 2.50MHz   -> real 2.00MHz  
Jtag frequency : requested 10.00MHz  -> real 6.00MHz  
erase SRAM Done
Detail: 
Jedec ID          : 0b
memory type       : 40
memory capacity   : 16
EDID + CFD length : 0b
EDID              : 1640
CFD               : 
Detail: 
Jedec ID          : 0b
memory type       : 40
memory capacity   : 16
EDID + CFD length : 0b
EDID              : 1640
CFD               : 
RDSR : 00
WIP  : 0
WEL  : 0
BP   : 0
TB   : 0
SRWD : 0
flash chip unknown: use basic protection detection
Erasing: [==================================================] 100.00%
Done
Writing: [======================                            ] 42.09%Error: ftdi_read_data in mpsse_readError: ftdiWriting: [=======================                           ] 45.51%Error: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_readError: ftdi_read_data in mpsse_read
```

nand2mario, 2022-10-25:
I’ll try to reproduce when I get a chance.

nand2mario, 2022-10-26:
Assuming you are using the lite version, could you give me messages from the serial port (baud rate 921600)? If you have a led module (I use the Muselab PMOD-LED), then it would be even better if you can give me what it shows when plugged into the bottom-right pmod socket.

nand2mario, 2022-10-26:
Does nestang work when you write it to SRAM?

Chandler-Kluser, 2022-10-28:
Yes, only in SRAM it is possible to flash, the output:

```
openFPGALoader -b tangprimer20k nestang-0.4-lite.fs 
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Parse file Parse nestang-0.4-lite.fs: 
Done
DONE
Jtag frequency : requested 2.50MHz   -> real 2.00MHz  
erase SRAM Done
Flash SRAM: [==================================================] 100.00%
Done
```

nand2mario, 2022-10-28:
I was able to reproduce this problem. With latest openFPGALoader on Ubuntu 20.04, SRAM write is ok but write to flash failed.

A quick work-around is to use Gowin's programmer. I have been using that (1.9.8.08) and it works in both SRAM and flash mode. 

Does openfpgaloader work for you in flash mode for other .fs files? I'm wondering if it's something in nestang that triggered the problem.

Chandler-Kluser, 2022-10-28:
I believe I have bricked the external SPI flash from my first board because it has never worked again.

I have try to solder another chip on the board, but the pads came over :(

the second one as soon as I detected the same issue, I have stopped everything so it is running fine!

I am pretty sure there is something with the openFPGALoader and the bitstream file generated, but I cannot identify what is going on.

nand2mario, 2022-10-29:
Just a bit of more information, the previous version nesting-0.3-lite.fs also does not work in openFPGALoader flash mode.

Chandler-Kluser, 2022-11-13:
@nand2mario I am facing issues to flash the HDMI Dock version to Tang Primer 20k too, I have tried to use GoWin Official Programmer and openFPGALoader.

I have tried to flash the 0.4.0 dock version release and the 0.4.0 built from source in my PC but both have not worked....

The GoWin Official Programmer flashes the SPI but It stucks in memory verification, did you have some problems with that?

nand2mario, 2023-05-20:
Close since it is an Sipeed issue.

closed 2023-05-20 by nand2mario