nand2mario/nestang #3: loader does not work, even with redistributables

jon-nielsen, 2022-12-16:
Hello 

Thank you for creating this project.

Unfortunately loader.exe does not work with the linked redistributable. After installing, several are still missing, including msvcp140d.dll

A quick search on Google informs me that this is a debug version not included in any release redistributable.

Please advise

Thank you

Chandler-Kluser, 2022-12-18:
Thanks for your feedback

I did not have any issue by compiling and running loader.exe in my PC

Can you, please, share with us your specs?

If you have more details it will be helpful too!

jon-nielsen, 2022-12-18:
Thanks for the response. 

My specs are Windows 10 x64, fully updated, Gowin drivers and programmer installed, and using the 4.0 lite firmware flashed to the Tang board through the JTAG board.

As far as I can tell the loader.exe included with the release packages is a debug binary. So the VC redists available for download do not contain the debug versions of those DLLs.

Anyway. I spent some more time on this yesterday and decided to try and build loader from source using my main PC. Also Windows 10 x64. Updated Visual Studio to latest version (compiler v19.34 I believe), and was able to build project by specifying C++17 in the solution properties.

But only debug target worked. Since I'll be loading from other computers, I wanted to build release version.

There was a problem however with an import with release target, joyGetPosEx specifically. To get around this I downloaded SFML and included it in project like so: https://www.sfml-dev.org/tutorials/2.3/start-vc.php

With this I was able to build release version. I'm not sure this is the right import however.

Next problem is that the games do not actually start after loading has completed, and I'm not sure if this is due to any problem with my compiled loader.exe, the controller, or the tang board itself. I may create an issue about that if I'm able to better describe the problem.

Chandler-Kluser, 2022-12-18:
I am glad to know you were able to fully build the loader.exe in your PC.

In my computer I was only able to build the Debug Target with my latest MSVC as you were. I am not sure, but I believe the project was made for Windows only with the Debug Target.

It would be a nice thing to include in the project to do list.

jon-nielsen, 2022-12-19:
I agree. The ideal outcome of this issue ticket is to change project settings to allow build of release artifact - and to package this with future releases.

closed 2023-08-10 by nand2mario