nand2mario/nestang #5: issue burning FPGA

rabarar, 2023-03-18:
I have a dock and accidentally burned the lite core - and now I can't burn the dock core.

Have I fried it or his it in a state that can be reset?
When I run openFPGALoad --detect I see the board
```
% sudo openFPGALoader --detect                          
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
index 0:
	idcode 0x81b
	manufacturer Gowin
	family GW2A
	model  GW2A(R)-18(C)
	irlength 8
```

but when I attempt to load the core, I get:

```
% sudo openFPGALoader -b tangprimer20k -f nestang-0.4-dock.fs 
write to flash
Jtag frequency : requested 6.00MHz   -> real 6.00MHz  
Parse file Parse nestang-0.4-dock.fs: 
Done
DONE
Jtag frequency : requested 2.50MHz   -> real 2.00MHz  
Jtag frequency : requested 10.00MHz  -> real 6.00MHz  
erase SRAM Done
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
EDID + CFD length : ff
EDID              : ffff
CFD               : 
Detail: 
Jedec ID          : ff
memory type       : ff
memory capacity   : ff
EDID + CFD length : ff
EDID              : ffff
CFD               : 
RDSR : ff
WIP  : 1
WEL  : 1
BP   : f
TB   : 1
SRWD : 1
flash chip unknown: use basic protection detection
unlock blocks
Error: block protection is set
       can't unlock without --unprotect-flash
Error: Failed to program FPGA: Error: write to flash failed
```

any thoughts?

nand2mario, 2023-03-19:
Try something like "loader pacman.nes".

The loader takes the binary rom (ines format), not the .hex file.

rabarar, 2023-03-19:
okay - was able to load a rom! see the game running, but there's no response to the controllers. I'm using the NES original controllers

rabarar, 2023-03-19:
UPDATE2: gamepad - 

so `lsusb` shows `Personal Communication Systems, Inc. SNES Gamepad` which I don't see in the gist sourced in the code... is that an issue?

```https://gist.github.com/nondebug/aec93dff7f0f1969f4cc2291b24a3171```

zf3, 2023-03-19:
Sound: Are you on a TV or monitor? I ran into the no sound issue on a Sony TV too. I had a fix for that checked into master (unfortunately the code is in flux now).
 
Gamepad: that's probably the issue. Could you post the relevant lsusb -v part? That should show the vendor and product id.

nand2mario, 2023-03-19:
Sound: Are you on a TV or monitor? I ran into the no sound issue on a Sony TV too. I had a fix for that checked into master (unfortunately the code is in flux now).

Gamepad: that's probably the issue. Could you post the relevant lsusb -v part? That should show the vendor and product id.

rabarar, 2023-03-19:
I’m using a Sharp4K Monitor.

Here’s the lsusb verbose output:

Bus 007 Device 015: ID 0810:e501 Personal Communication Systems, Inc. SNES Gamepad
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0810 Personal Communication Systems, Inc.
  idProduct          0xe501 SNES Gamepad
  bcdDevice            1.06
  iManufacturer           0 
  iProduct                2 usb gamepad           
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      89
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10

nand2mario, 2023-03-19:
Your device 0810:e501 is actually in the gamepad list.  So it seems to be something else.

Try pass -v to loader. It will print everything the loader passes to the fpga. Check if there are packets when you press controller buttons.

rabarar, 2023-03-19:
Hmm,

When I press any button, the only output I see is the last two lines of hex — same regardless of which buttons I press:

mario.nes transmitted over /dev/ttyUSB1 at baudrate 921600.
Found 1 gamepads
Opening gamepad /dev/input/event16
bf 40 01 00 
be 41 01 00

rabarar, 2023-03-19:
So I looked at the code - saw you had debug printfs in util.cpp on 389,390 for events - commented them out and reran it

Getting inputs from the controller…

Event: time 1679238400.416185, type: 3, code: 0, value: 0
Event: time 1679238400.624183, type: 3, code: 0, value: 127
Event: time 1679238402.392172, type: 3, code: 0, value: 0
Event: time 1679238402.520171, type: 3, code: 0, value: 127
Event: time 1679238403.008169, type: 3, code: 0, value: 0
Event: time 1679238403.048214, type: 3, code: 1, value: 255
Event: time 1679238403.096175, type: 3, code: 1, value: 127
Event: time 1679238403.176174, type: 3, code: 0, value: 127
Event: time 1679238403.504182, type: 3, code: 0, value: 0
Event: time 1679238404.448175, type: 3, code: 0, value: 127
Event: time 1679238404.904162, type: 3, code: 0, value: 0
Event: time 1679238405.480161, type: 3, code: 0, value: 127
Event: time 1679238406.176157, type: 3, code: 0, value: 255
Event: time 1679238406.224157, type: 3, code: 1, value: 0

So buttons are pressing and working … seems like it’s not going to the FPGA..

rabarar, 2023-03-19:
UPDATE:

Some points of clarification:
1. I cannot get the Gowin software to recognize the FTDI USB port and use it to program... solely using openFPGALoader
2. using OpenFPGALoader to load flash fails consistently
3. using OpenFPGALoader to load SRAM works but takes more than a few attempts before it either loads 100% and reboots after loading to accept roms correctly from the loader

not sure if this is expected behavior or if this is an indication of a configurations/hw issue with the TangPrimer20K

Chandler-Kluser, 2023-03-19:
maybe [this link](https://github.com/trabucayre/openFPGALoader/issues/251) can help you...

rabarar, 2023-03-19:
Read it - tried it - no dice … the issue is with the FGPA and the UART via the FTDI — loads the roms but that’s it…

Not sure why thought I can load the rom, load the core (in SRAM but not FLASH) but not have the loader talk to the USB/UART …

nand2mario, 2023-03-20:
It looks like a defective board or BL702 firmware bug to me. I have 3 tang primer 20k and the Gowin programmer works on all 3. Maybe Sipeed can help?

rabarar, 2023-03-20:
Is there a way to upgrade the firmware on the board?

Are you using the Linux or Windows version of the Gowin software?

nand2mario, 2023-03-20:
I'm using windows Gowin programmer and openFPGALoader on Windows/Linux and Mac.

I haven't upgraded the firmware on this board.  But afaik the BL702 firmware is closely related to this: https://github.com/sipeed/RV-Debugger-BL702

Try asking sipeed in places like here: https://github.com/sipeed/sipeed_wiki/issues

CoreRasurae, 2023-06-04:
I had this issue. My Tang Primer 20K Dock would always hang during the programming with openFPGALoader on Linux. I have reflashed the firmware of the BL702 and now I am both able to flash and load to load into SRAM without hanging, except that now the serial UART with the FPGA is not working. No communication at all. I have temporarily circumvented the problem by assigning a PMOD connector to an external USB to UART adapter and it not only works, as it is much more reliable loading games through this externall adapter, than going through the BL702.

rabarar, 2023-07-14:
How do I reflash the firmware on the BL702?

> I had this issue. My Tang Primer 20K Dock would always hang during the programming with openFPGALoader on Linux. I have reflashed the firmware of the BL702 and now I am both able to flash and load [...]

Chandler-Kluser, 2023-07-14:
> I had this issue. My Tang Primer 20K Dock would always hang during the programming with openFPGALoader on Linux. I have reflashed the firmware of the BL702 and now I am both able to flash and load [...]

this could be in a wiki page, could you provide us more details?

CoreRasurae, 2023-07-15:
The procedure to generate and flash the BL702 usb2uartjtag is as follows:

```git clone https://github.com/sipeed/RV-Debugger-BL702
cd RV-Debugger-BL702
git submodule init
git submodule update

cd bl_mcu_sdk
make APP=usb2uartjtag APP_DIR=../app CHIP=bl702 INTERFACE=/dev/ttyACM0

cd tools/bflb_flash_tool/
./bflb_mcu_tool --chipname=bl702 --port=/dev/ttyACM0 --xtal=32M --firmware="../../out/app/usb2uartjtag/usb2uartjtag_bl702.bin"
```

rabarar, 2023-07-15:
What OS/platform are you building and running this on?

CoreRasurae, 2023-07-15:
Ubuntu 20.04 LTS

rabarar, 2023-07-15:
What are the riscv toolchain dependencies used for the build on Ubuntu?

CoreRasurae, 2023-07-15:
I don't remember where I got it from, but it came from a zip file named:
toolchain_riscv_thead_linux64.zip

rabarar, 2023-07-15:
Cool - I’ll hunt it down…

rabarar, 2023-07-15:
In case anyone else is looking for it:stnolting/riscv-gcc-prebuilt: 📦 Prebuilt RISC-V GCC toolchains for x64 Linux.github.com

rabarar, 2023-07-15:
Any chance you can post a link to usb2uartjtag_bl702.bin

I can’t get it to build…

CoreRasurae, 2023-07-17:
Maybe you forgot to update cmake to 3.26.4, it is mandatory on my Ubuntu to make a local installation of that cmake version.
With that added to the PATH and riscv toolchain, it gets straightforward.

rabarar, 2023-07-18:
so burning fails... am I missing something??

```
$ sudo ./bflb_mcu_tool --chipname=bl702 --port=/dev/ttyACM0 --xtal=32M --firmware=/home/robert/usb2uartjtag_bl702/usb2uartjtag_bl702.bin 
[19:56:37.260] - ==================================================
[19:56:37.260] - Chip name is bl702
[19:56:37.261] - Serial port is /dev/ttyACM0
[19:56:37.261] - Baudrate is 115200
[19:56:37.261] - Firmware is /home/robert/usb2uartjtag_bl702/usb2uartjtag_bl702.bin
[19:56:37.261] - Default flash clock is 72M
[19:56:37.261] - Default pll clock is 144M
[19:56:37.261] - ==================================================
[19:56:37.267] - Update flash cfg finished
[19:56:37.268] - EFUSE_CFG
[19:56:37.268] - BOOTHEADER_CFG
[19:56:37.271] - Create bootheader using /home/robert/src/NES/RV-Debugger-BL702/firmware/bl_mcu_sdk/tools/bflb_flash_tool/chips/bl702/img_create_mcu/bootheader_cfg.ini
[19:56:37.271] - Updating data according to </home/robert/src/NES/RV-Debugger-BL702/firmware/bl_mcu_sdk/tools/bflb_flash_tool/chips/bl702/img_create_mcu/bootheader_cfg.ini[BOOTHEADER_CFG]>
[19:56:37.273] - Created file len:176
[19:56:37.275] - Image create path: /home/robert/src/NES/RV-Debugger-BL702/firmware/bl_mcu_sdk/tools/bflb_flash_tool/chips/bl702/img_create_mcu
[19:56:37.275] - Config file: /home/robert/src/NES/RV-Debugger-BL702/firmware/bl_mcu_sdk/tools/bflb_flash_tool/chips/bl702/img_create_mcu/img_create_cfg.ini
[19:56:37.275] - Image hash is b'7bf22d2b8644ead66173b83728f792f9efaad2d71bb152646e9cb8b96462a053'
[19:56:37.275] - Header crc: b'233e4063'
[19:56:37.275] - Write flash img
[19:56:37.276] - ========= eflash loader config =========
[19:56:37.283] - Version: bflb_eflash_loader_v2.1.2
[19:56:37.283] - Program Start
[19:56:37.283] - ========= eflash loader cmd arguments =========
[19:56:37.283] - Config file: /home/robert/src/NES/RV-Debugger-BL702/firmware/bl_mcu_sdk/tools/bflb_flash_tool/chips/bl702/eflash_loader/eflash_loader_cfg.ini
[19:56:37.284] - serial port is /dev/ttyACM0
[19:56:37.284] - chiptype: bl702
[19:56:37.284] - ========= Interface is uart =========
[19:56:37.284] - com speed: 115200
[19:56:37.284] - Eflash load helper file: /home/robert/src/NES/RV-Debugger-BL702/firmware/bl_mcu_sdk/tools/bflb_flash_tool/chips/bl702/eflash_loader/eflash_loader_32m.bin
[19:56:37.284] - Eflash load helper file: /home/robert/src/NES/RV-Debugger-BL702/firmware/bl_mcu_sdk/tools/bflb_flash_tool/chips/bl702/eflash_loader/eflash_loader_32m.bin
[19:56:37.284] - ========= load eflash_loader.bin =========
[19:56:37.284] - Load eflash_loader.bin via uart
[19:56:37.284] - ========= image load =========
[19:56:37.291] - usb serial port
[19:56:37.341] - clean buf
[19:56:37.341] - send sync
[19:56:37.443] - ack is b''
[19:56:37.443] - retry
[19:56:37.443] - usb serial port
[19:56:37.494] - clean buf
[19:56:37.494] - send sync
[19:56:37.696] - ack is b'4f4b'
[19:56:37.726] - shake hand success
[19:56:37.736] - get_boot_info
[19:56:38.843] - FL0102 unknown
[19:56:38.843] - Not ack OK
[19:56:38.843] - FL0102 unknown
[19:56:38.843] - result: FL0102 unknown
[19:56:38.844] - fail
[19:56:38.844] - Img load fail
[19:56:38.844] - Load helper bin time cost(ms): 1559.5830078125
[19:56:38.844] - {"ErrorCode": "0003","ErrorMsg":"BFLB LOAD HELP BIN FAIL"}
[19:56:38.844] - Burn Retry
[19:56:38.844] - 0
[19:56:38.844] - Burn return with retry fail
```

rabarar, 2023-07-18:
> Here it goes the compiled usb2uartjtag:
> [usb2uartjtag_bl702.zip](https://github.com/nand2mario/nestang/files/12071540/usb2uartjtag_bl702.zip)

installed `snap` version of `cmake`  and rebuilt without issue

closed 2023-08-10 by nand2mario