nand2mario/nestang #6: USB gamepad support

nand2mario, 2023-05-20:
It may be possible to add USB gamepad support through hi631's usb-ukp technique: https://qiita.com/hi631/items/4f263ca676e4be14b9f8 and http://kwhr0.g2.xrea.com/hard/pc8001.html

We probably need both HID and XINPUT.

mopplayer, 2023-06-03:
It is OK to support the DS2 gamepad.
I use a USB Keyboard with my Tang Primer (Anlogic), just for reusing my old keyboard and do not need to buy an additional gamepad.
And if you do not use it, you can insert it into a computer for reusing.

Chandler-Kluser, 2023-06-03:
@nand2mario I will try to work out a way to make it work with a generic USB gamepad, it is a nice feature to have in the project

nand2mario, 2023-06-04:
> @nand2mario I will try to work out a way to make it work with a generic USB gamepad, it is a nice feature to have in the project

That'll be great.

Chandler-Kluser, 2023-06-11:
@nand2mario the USB implementation in [image] was made with UKP - USB Keyboard Processor (only works in low speed USB devices, which I do not have anyone here to test, although I could build and test the TG9 NES core successfully).

The good part is that it has been already tested by @hi631 and it takes ~430 LUTs:

[image]

The bad part is about devices compatibility, I do not believe it should be a nice idea to use this way.

I believe [image] could help us better with UTMI implementation, taking about ~580 LUTs:

[image]

what do you think about?

nand2mario, 2023-06-11:
@Chandler-Kluser Thanks for the pointer and I'll take a look at core_usb_host.

Very quickly it seems to use AXI_lite and requires a CPU to function, which the nano 20K does not have. USB is designed with software drivers in mind so making a usb host with FPGA is kinda hard.

Chandler-Kluser, 2023-06-11:
maybe a cheap microcontroller is a better option, I know the intention is to make the board to do all the stuff...

Chandler-Kluser, 2023-06-15:
I know its not intended to be done but I have connected an ESP32 with a blueretro firmware in [FPGANES](https://github.com/hi631/tang-nano-9K) project in Tang Nano 9k exchanging the UKP USB module  DS2 interface and it worked pretty well in TG9K.

The pros of doing that are:

- the vast majority of bluetooth joypads have XInput interface
- incorporation of all supported joypads from [blueretro](https://github.com/darthcloud/BlueRetro) project
- less utilization of LUTs comparing with a USB Host module to drive Xinput devices
- the user will still have the option of connecting a real DS2 controller
- very cheap option, since esp32 has very low cost

It has its drawbacks:

- it will require a microcontroller (ESP32) in the project (I know for some people it is not a goal)
- only XInput bluetooth or real DS2 joypads will work, still no USB
- it will require some soldering and DIY tasks (unless someone makes an integrated hardware for that)

what do you think, @nand2mario ?

Chandler-Kluser, 2023-06-15:
another option is using [tinyusb](https://github.com/hathach/tinyusb) lib with some cheap microcontroller to get the desired USB interface.

in my opinion, adding this kind of functionality directly in FPGA will take great number of FPGA resources.

wolfshow, 2023-06-20:
> maybe a cheap microcontroller is a better option, I know the intention is to make the board to do all the stuff and USB Gamepad support would be such a nice feature...
> 
> but thinking about your suggestion, ukp is also a cpu written in verilog so I do not see any alternative for that

https://github.com/nand2mario/nestang/assets/3775235/058f10be-c9ce-4109-b957-c255db68bfc9

Chandler-Kluser, 2023-06-25:
I have posted a [project with bluetooth support](https://www.youtube.com/watch?v=wjiUXOrpm1Q) by merging [hi631](https://github.com/hi631/tang-nano-9K) NES core with the DS2 from [nestang](https://github.com/nand2mario/nestang) project.

nand2mario, 2023-08-16:
There's now a working first version of USB gamepad support on the [usb_gamepad branch](https://github.com/nand2mario/nestang/tree/usb_gamepad). At least it works with my gamepad.

Connect D+ to pin 41 and D- to 42, and add two 15k pull-down resistors as shown below.

[image]

Todo: gamepad type detection as apparently there are many different kinds; and 2nd player, and autofire.

nand2mario, 2023-08-19:
This is just released in v0.6. Thanks @Chandler-Kluser for your help.

closed 2023-08-19 by nand2mario