nand2mario/nestang #7: Bitstream loading through BL616 on Nano 20K

nand2mario, 2023-05-20:
In theory, BL616 can read bitstream from SD card or USB and program the FPGA, allowing switching between cores.

Some discussion here: https://www.reddit.com/r/GowinFPGA/comments/11u5vlf/usb_cdcserial_repl_for_pikapython/

Issues:
- The BL616 doesn't have enough memory to hold the whole fs image (8MB). 
- BL616 is wired as a USB device (CC pins tied to ground). So it cannot act as a USB host even if the sdk allows it.

harbaum, 2024-01-09:
The BL616 can act as a USB host. See https://www.youtube.com/shorts/bP5gK3nmv-o

harbaum, 2024-01-09:
The fs image is only ~700k. The .fs files are ASCII and thus way bigger. But 700k is still too much for the BL616. I am considering using an external SPIU or I²C RAM.