nand2mario/nestang #8: Great job

mopplayer, 2023-06-03:
First of all, I thank @nand2mario and @Chandler-Kluser for contributing those efforts to the NES platform.

However, there are many copied and reworked NES platforms nowadays.

For example, MAME (and many soft emulators) and MIST (using ALTERA Cyclone III), MiSTer (using ALTERA Cyclone V SOC).
It may not be so beneficial to do those tasks again.

Moreover, there are so many copied and cracked NES ASIC platforms in China. The product runs just like the original NES hardware, and does not need to redo a China FPGA platform.

I have also taken some time to do those tasks on Tang Primer (Anlogic), but have seen that it is not so useful for marketing and contributing. 

Yeah, I realized FPGAs should do like its advantage: computing and real-time, flexible.

This comment does not mean giving up something, but should do better.

You could see my old project video, this project was abandoned:
https://www.youtube.com/watch?v=OIXi4LuuK1E

P.S. 
Creating a game menu is a good idea, but using a python script to create a game image is not so convenient if you have no computer around...

I am so lazy, just put game1.nes, game2.nes,... game12.nes to SD card, it represents F1~F12 key respectively. Press and start. You could put those files to a SD card with your smartphone.

Chandler-Kluser, 2023-06-03:
Can you share your HDL designs to open source community? Thanks in advance!

nand2mario, 2023-06-04:
Thanks @mopplayer for sharing your thoughts and your project.

I chose Sipeed/Gowin at the beginning because of two reasons. First, the boards are really low-cost, in good quality and more importantly come with reasonably good dev tools. Second, Sipeed iterates quickly with their boards and seem to actually listen to developers. SDRAM was added to Nano 20K based on our feedback.

The overall aims is to develop enough basics / infrastructure, so that we have a low-cost, easy-to-setup FPGA retro game platform for myself and others to enjoy. NES is both an end and also a means to support other 8-bit and possibly 16-bit platforms.

FAT32-based SD card loading is the goal. For that we need to figure our FAT parsing. I think the current menu system is okay but it is not super easy to set up the card as you said. It should be relatively easy to develop a Mac/Windows program for filling the card.

mopplayer, 2023-06-11:
Hi @nand2mario, 

According to my knowledge, the Nano 20K just selects a GW2A"R"-18, just like Anlogic EG4"S"20 in previous Tang Primer. They both embedded a SD"R"AM ("S"DRAM). They both have a product line for a long time.

To design a board is not a hard task, which even is low cost in China. The problem is, support is a common weakend point in China. So that's why it is so cheap, just a law of marketing to survive.  Iterating boards quickly is not a good idea, just pushes a new cheap board to market to earn incoming and less support. That's common sense for those manufacturing development board companies in China. I have been given or bought many boards from China for a long time.

Yeah, a cheap FPGA retro game platform. However, I have motioned that there are many cracked and reworked ASIC platforms in China, which the price even is low to less than 10 USD (aliexpress searched). A player wants to play a NES game with real hardware, which will he buy? 

Might be other old hardwares have some contributions on education, like Z80, is a good direction. Mac/Windows programs still need a computer around.

@Chandler-Kluser If my work and project is finished for a stage, may I refine that old code for considering open source.

closed 2023-08-10 by nand2mario