nand2mario/nestang #11: segmenation fault

wolfshow, 2023-06-09:
why do I get segmentation fault when loading the newly released 0.5, but no problem with 0.5rc

Chandler-Kluser, 2023-06-09:
can you provide more details about the build configuration?

wolfshow, 2023-06-09:
> can you provide more details about the build configuration?

[image]

wolfshow, 2023-06-09:
> Can you provide more details like:
> 
> * Were you able to write it into SRAM?
    I haven't tried this yet because I am able to write 0.5rc.

> * What openFPGALoader build did you use?
    The latest build.

> * Did you try to flash using Gowin Programmer?
    No, I did not use Gowin Programmer.

Chandler-Kluser, 2023-06-09:
I believe the issue can be in openFPGALoader, I have already faced [image] in Tang Primer 20k by writing its flash memory and did not have any by using the Official Programmer.

Another detail could be any kind of bitstream generation by using nextpnr, for instance. But I don't know if you placed and routed with gw_sh or nextpnr.

I will try to reproduce your steps here.

wolfshow, 2023-06-10:
> I believe the issue can be in openFPGALoader, I have already faced [image] in Tang Primer 20k by writing its flash memory and did not have any by using the Official Programmer.
>
> Another detail could [...]

Thanks! For the official GOWIN programmer, do I need to use version 1.99 with commercial license? or 1.98 is fine?

Chandler-Kluser, 2023-06-10:
the Educational Edition is fine, because there is support for Tang Nano 20k

closed 2023-08-10 by nand2mario