nand2mario/nestang #12: Standard nes controller

siffland, 2023-06-09:
Is there a way to have an alternate configuration at compile time for a standard 7 pin NES controller port to be wired to the nano20k instead of the PS2 ones that come with the kit?  Besides the 5 volt and ground there are just the other 5 pins to deal with.  It would make it a bit more authentic feeling.......or the homebrew analogue NT people can still purchase and afford :) .  I do not know if that is a huge effort in code, but would think not (I could be wrong).

Chandler-Kluser, 2023-06-09:
I am actually working in a [image] for USB Gamepads. A NES Gamepad Interface I believe that would be a lot simpler to implement, if you want to help us :)

wolfshow, 2023-06-10:
> I am actually working in a [image] for USB Gamepads. A NES Gamepad Interface I believe that would be a lot simpler to implement, if you want to help us :)

I have some ATmega32u4 driver for USB gamepad. https://github.com/wolfshow/gamepad

Chandler-Kluser, 2023-06-10:
thanks, I will look for that!!

nand2mario, 2023-06-11:
There's a verilog NES controller interface module: https://github.com/michael-swan/NES-Controller-SIPO 
And also for SNES controller: https://www.fpgalover.com/ip-cores/snes-controller-core.

I don't quite have time to look at this right now. So pull requests are welcome.

fjpolo, 2024-01-30:
[NESGamepad](https://github.com/nand2mario/nestang/pull/49) module will support NES origina joysticks #1 and #2 and has support for 8BitDo NES controllers

fpolomimi, 2024-03-20:
@nand2mario [#52](https://github.com/nand2mario/nestang/pull/52) adds OEM gamepad feature to NESTang_nano20k. I never implemented nor tested in in NESTang_primer25k since I don't own a board. 

This issue could be kept open to track until Primer25k supports OEM, or be closed and add a new issue for Primer25k only

nand2mario, 2024-05-19:
Thanks for everyone's help. The new NES/SNES controller support is ported from snestang. Could you test the attached test build. I've tested SNES controllers and they work fine for both primer25k. Help with NES controllers are needed as my pair is currently on the way. Also I haven't got around to test on nano 20k yet.

[nestang-0.10-test-snes-nes-controllers.zip](https://github.com/nand2mario/nestang/files/15368794/nestang-0.10-test-snes-nes-controllers.zip)

nand2mario, 2024-05-19:
It's probably the autofire misbehaving. Let me fix that.

nand2mario, 2024-05-19:
According to https://tresi.github.io/nes/, for the NES controller, data line will be 0 after 8 clock cycles. As signal is negative true, this will indicate the other buttons to be "pressed" if we read this as a SNES controller signal. The current code uses SNES A and X buttons as auto-fire buttons for NES. Therefore A and B becomes repeatedly pressed and unpressed.

To fix this, we can detect between SNES and NES controller (https://www.nesdev.org/wiki/Controller_detection), and apply autofire only for SNES controllers.

fpolomimi, 2024-05-22:
Sorry I'm late to the show, I don't know why I'm subscribed with my work
email instead of my personal one ***@***.***). I'll see what I can do
whenever I have some time to test/debug it.

-- 
Franco J. Polo
Embedded Engineer
Phone: +49 (0)30 20236420-0

nand2mario, 2024-05-23:
#74 is a duplicate of this. I've just merged a fix by @nikitalita. Let me know if it works for you.