nand2mario/nestang #13: 2nd player controller button mapping

siffland, 2023-06-10:
I was playing Super Mario Bros with 2 players, utilizing the Sipeed ps2 controller adapters (Both wired on pins 15-20).  The first controller works fine.  With the second controller the d-pad left, B and A buttons work as expected.  The d-pad up runs luigi to the right and d-pad right and down do nothing.  

Contra the start button is up and the right button is right and down combined for the second player.

I did swap the controllers and the adapter boards and all work fine on the player 1, it is only player 2

I also created a wiki page on here for connecting the adapters on the breadboard (with a picture).  I believe they are hooked up correctly and this is an issue with the firmware.  I am using Nestang v0.5.

Chandler-Kluser, 2023-06-10:
@nand2mario

nand2mario, 2023-06-11:
Thanks for contributing the wiki pages. Let me take a look at the 2nd controller issue.

nand2mario, 2023-06-24:
I can reproduce this on my side. If someone has time, help will be appreciated in debugging this.

somhi, 2023-08-10:
Following is working: In nestang.cst file I changed the pin locations as you can see below.   That implies that the second joystick should be connected through Dupont wires to the corresponding pins.
´´´
// Joystick 2
IO_LOC "joystick_clk2" 28;
IO_PORT "joystick_clk2" PULL_MODE=NONE IO_TYPE=LVCMOS33;
IO_LOC "joystick_mosi2" 52;
IO_PORT "joystick_mosi2" PULL_MODE=NONE IO_TYPE=LVCMOS33;
IO_LOC "joystick_miso2" 53;
IO_PORT "joystick_miso2" PULL_MODE=UP IO_TYPE=LVCMOS33;
IO_LOC "joystick_cs2"  27;
IO_PORT "joystick_cs2" PULL_MODE=NONE IO_TYPE=LVCMOS33;
´´´

A much better solution would be to adapt the dualshock controller to handle two gamepads, and so only another CS pin will be necessary to control both joysticks. I don't have the level to do that. Hope someone could work on it. Help Wanted :)

For reference some of the test I made are in this duplicated [issue](https://github.com/nand2mario/nestang/issues/21)