nand2mario/nestang #18: Controller slot?

burntbyhellfire, 2023-07-16:
Getting a 60-pin card edge connector for the Famicom is pretty easy, and I have a few.  The Sipeed Tang has enough GPIOs that it should be able to handle it.  Has anyone considered an option of building something like this with a proper cartridge connector and controller ports?  I wouldn't be opposed to designing a PCB myself for the tang to connect into that can make full use of a new PCB designed to fit the standard Famicom motherboard footprint for original shell compatibility.

windship, 2023-07-19:
Absolutely agreed!

muramasatheninja, 2023-09-17:
Very interested in this.

nand2mario, 2023-09-18:
The nano 20k does not have enough pins for this. I count roughly 40 pins including the RGB-LCD socket.

Tang Primer 20K has enough pins. But it does not have SDRAM.

There's a GW5A-based nano coming. We can look at that once it comes out.

muramasatheninja, 2023-10-18:
Would it be possible to do a 20k primer version of the tang? I assume since it has enough pins we could have the cartridge but would have to give up rom loading from SDRAM. (or maybe sdram could be added to a board) Could be worth pursuing for some users. I would love to make board for a version like this.

I also ordered a Tang Mega 138k to hopefully try with as well.

nand2mario, 2023-10-18:
There's also an upcoming primer 25k SoM. https://wiki.sipeed.com/hardware/zh/tang/tang-primer-25k/primer-25k.html (seems the English page is not up yet)

nand2mario, 2023-10-23:
> Would it be possible to do a 20k primer version of the tang? I assume since it has enough pins we could have the cartridge but would have to give up rom loading from SDRAM. (or maybe sdram could be [...]

@muramasatheninja what is the best way to DM you or email? I'd like to discuss the board idea.

muramasatheninja, 2023-10-23:
> @muramasatheninja what is the best way to DM you or email? I'd like to discuss the board idea.

You can use my contact form on my website, or I already follow you on twitter you should be able to dm me there.
muramasaentertainment.com/contact-us/

Roet-Ivar, 2024-10-18:
I love this idead! The picture on the  famicom clone I'm sitting on right now is so bad I have to squint my eyes and pretend it's 80s. I'd love the idea to gut it (it looks very authentic btw) and put this in, a working cartridge connector would be amazing!
If we had USB support it could've been mounted as a flash drive from something like this:
https://github.com/digilogistist/The-Open-Source-USB-NES-Project?tab=readme-ov-file