nand2mario/nestang #21: Dualshock controller pin usage

somhi, 2023-08-07:
Being the Dualshock controller an SPI controller, it is using 8 GPIO pins but controlling the 2 CS lines should be enough with 5 pins. In MiSTer the PSX SNAC addon only uses 5 pins.  

For the kit distributed by Sipeed it is ok to connect the two gamepads on both sides of the tang nano 20k, but for a future carrier board I think only 5 pins would be sufficient.   I'm not aware if heavy modifications on the controller should be made to allow that.

somhi, 2023-08-09:
Results of pin testings to fins reliable pins for connecting Joystick #2 are really bad. I that side of the tangnano 20k I only found 4 pins which work.  
Not really sure what the problem can be. SPI frecuency is los 250 kHz.  I tested it in Pacman core but I believe that working config below will work in Nestang too.
Maybe someone could reproduce the tests and have an explanation of what is going on here.

What we really need is adapting the dualshock controller to handle two gamepads, and so only another CS pin will be necessary to control both joysticks.  I don't have the level to do that. Hope someone could work on it. 

```
// DUALSHOCK CONTROLLER PIN TESTING (3" DUPONT WIRES)
// FIXED CLK / CS PINS: 28/20
// MOSI / MISO  WORKS: 76/80, 53/52
// MOSI / MISO  NOT WORKING: 42/41, 56/54, 51/48, 51/55, 86/79, 72/71
// MOSI / MISO  WORKS BADLY 48/49
# IO_LOC "joystick_clk" 28;
# IO_PORT "joystick_clk" PULL_MODE=NONE IO_TYPE=LVCMOS33;
# IO_LOC "joystick_mosi" 53;  
# IO_PORT "joystick_mosi" PULL_MODE=NONE IO_TYPE=LVCMOS33;
# IO_LOC "joystick_miso" 52;
# IO_PORT "joystick_miso" PULL_MODE=UP IO_TYPE=LVCMOS33;
# IO_LOC "joystick_cs"  20;
# IO_PORT "joystick_cs" PULL_MODE=NONE IO_TYPE=LVCMOS33;

// DUALSHOCK CONTROLLER #2 WORKING PINOUT
IO_LOC "joystick_clk" 52;
IO_PORT "joystick_clk" PULL_MODE=NONE IO_TYPE=LVCMOS33;
IO_LOC "joystick_mosi" 53;  
IO_PORT "joystick_mosi" PULL_MODE=NONE IO_TYPE=LVCMOS33;
IO_LOC "joystick_miso" 80;
IO_PORT "joystick_miso" PULL_MODE=UP IO_TYPE=LVCMOS33;
IO_LOC "joystick_cs"  76;
IO_PORT "joystick_cs" PULL_MODE=NONE IO_TYPE=LVCMOS33;
```

nand2mario, 2023-08-10:
Thanks for doing the testing! I'll take a look.

nand2mario, 2023-08-10:
Duplicate of #13

closed 2023-08-10 by nand2mario