nand2mario/nestang #24: NES file loader

windship, 2023-08-19:
Currently I have to make games.img from my NES rom files collection but this is so uncomfortable. I have about 1,000 NES files but I can't make a games.img file including my all roms(because of max letter limit for one dos command). And I have to make another games.img file if I want to delete some roms from my SD card every time. Please enhance the loader that can load *.NES rom from SD card folder after selecting from the rom list. I don't know why games.img packing needed.

nand2mario, 2023-08-19:
I'm looking at file system support, now that USB gamepad is done in v0.6 (at least for a first version).

nand2mario, 2023-09-17:
FAT32 support is released in v0.7. Enjoy.

closed 2023-09-17 by nand2mario