nand2mario/nestang #28: HDMI and sound update from upstream

nand2mario, 2023-09-18:
Need to look into this:

There's an update to fix a major HDMI bug used in verilog code for NESTang.
It fixes data island offset bug that prevented many sinks from functioning properly.

Also there's a bug in sound generation using the old APU implementation. The waveforms miss the high pitch notes which is vivid in some games like SMB3 and the first TMNT on most screens. I even wrote some modifications on the original NES core to output sound via AUX and it has the same issue.

Mister NES core utilizes an updated system verilog module with more high-pitch friendly synth tables. The original core author intended to use these values for Nexys board with a different DAC type.

Octadot, 2023-09-21:
The HDMI bug issue is already solved:
Potential violation of the HDMI 12 pixel minimum control period..
https://github.com/hdl-util/hdmi/issues/43#issuecomment-1683013723

About the sound bug it's probably caused by the GW2AR-LV18 FPGA due to a possible lag at high frequencies when trying to access the APU lookup table for mixing. Splitting the table to square and others may solve the issue.
    tmp_a <= lookup1[in_a];
    tmp_b <= lookup2[in_b];

nand2mario, 2024-04-07:
Fixed in 0.9.

closed 2024-04-07 by nand2mario