nand2mario/nestang #30: Freezes/Resets on Roms Selection

s3anami, 2023-10-08:
When I select a rom the device freezes or sometimes resets itself. Not sure how to troubleshoot behavior

mariogazziro, 2023-10-10:
i notice that version 0.6 has a more stable menu system related to joystick selection without freezing.. i think that something in FAT sd card reader in 0.7.1 version compromise the menu system

s3anami, 2023-10-10:
Is there a way for me to provide logs? I couldnt get usb pad working either, tried the wireless xbox receiver mentioned in wiki. I wired in a esp32 for blueretro in ps2 mode and that works

nand2mario, 2023-10-10:
Xbox 360 controller is actually not supported. Currently only older (USB low-speed) controllers are working. Try something like this: https://www.aliexpress.us/item/3256802019916284.html

s3anami, 2023-10-13:
I tried a different sd card and this is resolved. Seems to be picky on sd card choice

closed 2023-10-13 by s3anami