nand2mario/nestang #31: USB Controller (DIY) Not Working [ Tang Nano 20K ]

rorrexatnys, 2023-10-10:
I followed the guide and wired my own USB port to pins 41 (D-) [white wire] and 42 (D+) [green wire], with a 15K resistor on each line going to GND,  5V power [red] and ground [black].  I've tried every USB gamepad I own, oldschool USB devices (NES USB adapter, Genesis classic USB controller, SNES USB adapter) and haven't had any response from menu.

-   I see a 1Hz poll on the data pin and I have voltage on controller, but no input response.
- I'm only using one USB port instead of the two shown, but I assume wiring only one USB doesn't affect operation?  I've also tried wiring to port 2 (pins 56/54) to no avail.
- I'm using nestang v0.6 which has USB gamepad support.
- I've tried swapping data pins to no avail.  Likewise tried with and without the pulldowns.

This is a great project, just not sure what I'm missing here for USB gamepad to work.  Perhaps I'm overlooking something.  Thanks in advance!

Chandler-Kluser, 2023-10-11:
I don't know if @nand2mario agrees, but I stick to the blueretro alternative (or using a cheap microcontroller to handle USB host through a library like tinyusb).

It is not a demerit to manage the peripherals to a microcontroller like MiSTer FPGA does, you save costly FPGA resources for things that are not necessarily related to the system processing besides the project cost will keep low since these devices are cheap.

If there is no issue about merging this idea to the project, I commit myself to do that.

nand2mario, 2023-10-13:
> I don't know if @nand2mario agrees, but I stick to the blueretro alternative (or using a cheap microcontroller to handle USB host through a library like tinyusb).
>
> It is not a demerit to manage the [...]

I think we should support both external IO controller (the blueretro approach you mentioned), and internal IO approach. The most important benefit of the blueretro approach is actually better compatibility with different peripherals like controllers. It is the approached used by MIST. The benefit of the internal approach, on the other hand, is ease-of-use. So we'll also keep that.

In short, I'll merge blueretro related code if it is clean enough.

BennehBoy, 2024-06-22:
Is USB controller via GPIO deprecated now?

Would it make sense to use an M0S dock similar to NanoMig, MySTeryNano etc?