nand2mario/nestang #32: Video Sync Issue

s3anami, 2023-10-13:
Occasionally the device will temporarily loose sync and other times completely lose video sync (requiring a reset) while everything else is still working ( I can hear audio and my characther seems to be moving by audio cues). I am using an LG C1 and have had sync issies with the Tang8k device before, although a fix was found there. The other thing I notices is that the Tang20k seems to get very hot and wondering if that could be related. I added a heatsink from a raspberrypi i had laying around to try and help there.

nand2mario, 2023-10-14:
Could be related to #28?

s3anami, 2023-10-14:
Not sure how or where to check to verify. TMNT was very high pitched in the intro

nand2mario, 2024-04-07:
Haven't seen this lately. So closing.

closed 2024-04-07 by nand2mario