nand2mario/nestang #33: Button combo to return back to game list?

Helder1981, 2023-10-14:
Is there a button combo to return back to the Game List? If not can there be one added so that you do not have to power the tango on and off.  Can the menu also cycle through the game list pages faster? It's a bit sluggish and with tons of games it takes a while to jump to the game you want or maybe have a menu option to jump to different letters like say jump to games that start with D.

I also got my prototype boards in and so far they are working as intended but a mistake was made with the PS2 ports where it has to be soldered on the bottom of the board vs the top but it works and will be fixed in version 2. I am waiting on an esp32 board to arrive to test Blue Retro with that and other controllers through Blue Retro.

USB ports work and only 1 of my generic usb controllers work (an N64 usb controler) every other controller I had is an Xinput controller and those do not work. Would be nice if support for that was added since many 3rd party controllers default to that mode like 8BitDo.

[image]
[image]
[image]

nand2mario, 2023-10-14:
Created #34 to track XInput / Xbox 360 support.

Helder1981, 2023-10-14:
Thanks for that, but any info on the button combo? Can we maybe add a button to one of the Tango GPIO to cause a Reset back to the menu?

mariogazziro, 2023-10-14:
the S1 button already does that.

Helder1981, 2023-10-14:
Thank you for that information, I did not see it anywhere or maybe I missed it when searching for it.

mariogazziro, 2023-10-14:
v0.5rc - The Tang Nano 20K Release

SD card support. Now games can be loaded through a MicroSD card. Use tools/nes2img.py to generate the SD card image. A simple menu system allows easy navigation. **Press S1 button on the board to return to menu**.

mariogazziro, 2023-10-14:
but i would appreciate an menu option in IO pins too, since is very hard to wire a external push button from S1 smd device

Helder1981, 2023-10-14:
> but i would appreciate an menu option in IO pins too, since is very hard to wire a external push button from S1 smd device

I just checked the schematic and unfortunately the GPIO 88 is not broken out to any of the header pins. But maybe some kind of code can be written to implement the the Menu Return with a controller button combo.

nand2mario, 2023-10-14:
> but i would appreciate an menu option in IO pins too, since is very hard to wire a external push button from S1 smd device

Button S1 and pin 48 are both resets.

mariogazziro, 2023-10-14:
well, its a reset in name purposes only. can be attached to any part of the cpu. but even acting like a reset, does the job to rertun to menu.

mariogazziro, 2023-10-14:
i dont think that a combo joystick button is necessary since we well need to add several LEs do filter de input and may compromises the response time of the joysticks inside nes hardware emulation.. a simple wire connecting this reset pin to gpio would help.

Helder1981, 2023-10-16:
> > but i would appreciate an menu option in IO pins too, since is very hard to wire a external push button from S1 smd device
> 
> Button S1 and pin 48 are both resets.

I just tested this and pin 48 to 3.3v did not do anything neither to GND so is it a different pin other than 48 that is also reset?

Helder1981, 2023-10-16:
> i dont think that a combo joystick button is necessary since we well need to add several LEs do filter de input and may compromises the response time of the joysticks inside nes hardware emulation.. a simple wire connecting this reset pin to gpio would help.

Once I get confirmation of what pin actually causes Reset and test it I will likely add code into the ESP32 (BlueRetro) so that Home Buttons on wireless controllers trigger that pin to cause a reset so this way it does not affect any of the LE on the FPGA.

nand2mario, 2023-10-17:
> Once I get confirmation of what pin actually causes Reset and test it I will likely add code into the ESP32 (BlueRetro) so that Home Buttons on wireless controllers trigger that pin to cause a reset [...]

That piece of code is actually committed after 0.7.1 so you'll need to build from source, or I will put up a test build when I get a chance.

Helder1981, 2023-10-17:
> That piece of code is actually committed after 0.7.1 so you'll need to build from source, or I will put up a test build when I get a chance.

That would be appreciated if you could.

fjpolo, 2023-10-17:
Would something like this work for a gamepad button combination? (let's say SELECT+START):

````reg sys_resetn = 0;
reg [7:0] reset_cnt = 255;      // reset for 255 cycles before start everything
always @(posedge clk) begin
    reset_cnt <= reset_cnt == 0 ? 0 : reset_cnt - 1;
    if (reset_cnt == 0)
        sys_resetn <= ~s1 & ~reset2 | ~(nes_btn[4] & nes_btn[5]);
end``

This would mean using some LEs, but not that many:
![image](https://github.com/nand2mario/nestang/assets/9550134/f51865b5-a630-4c16-92d7-66aa929b84ec)
![image](https://github.com/nand2mario/nestang/assets/9550134/1f1df9f5-b9ac-4a48-9b46-4bd68f35434f)

nand2mario, 2023-10-19:
> > That piece of code is actually committed after 0.7.1 so you'll need to build from source, or I will put up a test build when I get a chance.
> 
> That would be appreciated if you could.

Try this: https://github.com/nand2mario/nestang/releases/tag/v0.7.2rc

Helder1981, 2023-10-19:
> > > That piece of code is actually committed after 0.7.1 so you'll need to build from source, or I will put up a test build when I get a chance.
> > 
> > 
> > That would be appreciated if you could.
> 
> Try this: https://github.com/nand2mario/nestang/releases/tag/v0.7.2rc

That works perfect now. I'm going to see if I can make the home button enable a pin high on the esp32 and tie it to pin 48 to cause an exit of the game back to the menu.

fjpolo, 2023-12-01:
This branch should let the user use Start+Select to reset the system: [fjpolo/dev_ResetButtonCombo](https://github.com/fjpolo/nestang/tree/fjpolo/dev_ResetButtonCombo)

Bitstream:
[NESTang_0v8_SelectStartResetCombo.zip](https://github.com/nand2mario/nestang/files/13525219/NESTang_0v8_SelectStartResetCombo.zip)

⚠I use a cheap USB SNES controller, using both this combination and button reset bugs the USB gamepad. But I think it's because of the gamepad itself

fjpolo, 2024-01-25:
As an update: [NESGamepad](https://github.com/nand2mario/nestang/pulls) module will support origina NES joysticks, meaning it supports 8BitDo joysticks and its Home button (Select+Dn button combo)

fpolomimi, 2024-03-20:
@nand2mario I guess this issue can be closed

closed 2024-04-07 by nand2mario