nand2mario/nestang #36: Master System controller support

RyanMac84, 2023-10-23:
I’m no coder, more of a copy & paste warrior and have no experience of FPGA, but rather then trying to implement USB would it be possible using the GPIO of the Tango nano 20k to read the digital inputs from a Sega Master System controller? From a nostalgia perspective it’s very similar to the NES controller (rectangular pad with 2 buttons) but I think easier to read as it doesn’t use Data/Latch etc but instead each pinout is a direct connection to each button, therefore, no input lag either, unlike USB or what the BlueRetro would introduce. 

[image]

I hope that makes sense and is a plausible solution 🤷🏻‍♂️

fjpolo, 2023-12-01:
I don't know the SMS joystick and the electrical properties of the signals, but I believe it should be possible. Bear in mind that it probably works with 5V and NESTang works with 3V3 signals, so you'd probably need a logic shifter for high frequency sampling or you'd probably do with resistor voltage dividers for sampling @60-120Hz.

My approach would be to create a new module with a clock divider from 27MHz to 60-120Hz, and just sample every pin at posedge of this clock divider, and just output that register. I think that should be enough.

I don't know how the light sensor works, so I'd just keep that disconnected

fjpolo, 2023-12-01:
I think this branch should do it: [fjpolo/TangNano20k_SMSGamepap](https://github.com/fjpolo/nestang/tree/fjpolo/TangNano20k_SMSGamepap)

You'll get something like this if you place a scope:
[image]

And this would be the pinout:
[image]

Bitstream:
[nes.zip](https://github.com/nand2mario/nestang/files/13524853/nes.zip)

Can't test it myself since I don't own a SMS gamepad.

⚠Don't forget to convert your 5V logic to 3V3 logic!

fjpolo, 2023-12-06:
BTW @RyanMac84, NESTang parses controller as [Right Left Down Up Start Select B A] and this SMSGamepad module parses it as [0 0 Right Left Down Up  B A]. Here's a new version that parses it as [Right Left Down 0 0 Up  B A]:

[NESTang_Nano20k_0v8_SMSGamepad_20231206.zip](https://github.com/nand2mario/nestang/files/13579669/NESTang_Nano20k_0v8_SMSGamepad_20231206.zip)

I personally don't think it's gonna work that good, many NES games depend on Start and Select buttons, so idk how you'll gonna find a workaround for it

fjpolo, 2024-07-08:
@nand2mario can we close this one?

closed 2024-07-09 by nand2mario