nand2mario/nestang #41: ST7789 display support and original nes controller

jamclucas, 2023-11-16:
im trying to build a portable nes console with an original nes cotroller, need support for a st7789 display and pwn audio support with an original nes controller, my only option now is nescat but it will be cool to use nestang.

fjpolo, 2023-12-01:
I'm working on a module for the original NES gamepad, though it's only formally verified and simulated with Verilator, I haven't tested it with the controller since I'm waiting for the logic shifters: [fjpolo/TangNano20k_NESGamepad](https://github.com/fjpolo/nestang/tree/fjpolo/TangNano20k_NESGamepad)

⚠Keep in mind that you'll need to adapt 3V3 LVCMOS3V3 signals (latch and clock) from NESTang to 5V TTL from NES gamepad, and viceversa, 5V TTL signal (serial joystick data) to 3V§ LVCMOS3V3 from NESTang

⚠Also, still in development

fjpolo, 2024-01-30:
[NESGamepad](https://github.com/nand2mario/nestang/pull/49) supports original NES joysticks, no need of level shifters (good news!) and supports 8BitDo NES controllers