nand2mario/nestang #43: Outdated Verilator Build Instructions?

Seyviour, 2023-12-01:
Hello,

Before anything else, thank you for the effort you put into building this, and for sharing this with us!!

The instructions for simulating with Verilator in the `nestang/verilator` folder seem to be out of date. Specifically, the build instructions reference a `nes_tang20k.v` file that doesn't seem to exist. Here's an excerpt from `verilator/setup.sh`: 

```
ln -sf $DIR/src/mmu.v hdl
ln -sf $DIR/src/nes_tang20k.v hdl 
ln -sf $DIR/src/nes.v hdl
```

This (I believe) ultimately cascades into an error when I run `make sim`

```
### VERILATE ####
mkdir -p obj_dir
cd obj_dir; ln -sf ../game.nes.hex .; ln -sf ../hdl/src/*.txt .
# verilator --top-module NES_Tang20k --trace -cc -O3 -CFLAGS "-I/usr/include/SDL2 -D_REENTRANT -O3" -LDFLAGS "-lSDL2" hdl/*.v --exe sim_main.cpp
verilator --top-module NES_Tang20k -cc -O3 --exe -CFLAGS "-I/usr/include/SDL2 -D_REENTRANT -O3" -LDFLAGS "-lSDL2" hdl/*.v sim_main.cpp
%Error-PKGNODECL: hdl/memory_controller.v:12:8: Package/class 'configPackage' not found, and needs to be predeclared (IEEE 1800-2017 26.3)
   12 | import configPackage::*;
      |        ^~~~~~~~~~~~~
                  ... For error description see https://verilator.org/warn/PKGNODECL?v=5.019
%Error: hdl/memory_controller.v:12:8: Importing from missing package 'configPackage'
   12 | import configPackage::*;
      |        ^~~~~~~~~~~~~
%Error: Exiting due to 2 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
make: *** [Makefile:17: obj_dir/VNES_Tang20k.cpp] Error 1
```

I've tried replacing `nes_tang20k.v` with `nestang_top.sv` in `verilator/setup.sh` but that doesn't seem to work. 

TIA

nand2mario, 2023-12-04:
Thanks for catching this. Should be fixed now.

Also there's no more need to use setup.sh. Just `make` should be enough.

Seyviour, 2023-12-05:
Thank you very much! It works now.