nand2mario/nestang #47: Use the BL616/M0S for USB, file system and more

harbaum, 2024-01-09:
The Tang Nano 20k comes with a BL616 as a flasher. The BL616 is quite powerful and can easily handle USB host (mouse, keyboard, joysticks, mass storage) and is able to cope with fat and exfat file systems with long file names etc etc. I am using it in MiSTeryNano. Would it make sense to use it for nestang and/or snestang as well. This might even free up some resources in the FPGA.

If you don't want to re-flash the internal BL616 you could use an external M0S or the like.

Chandler-Kluser, 2024-01-09:
This actually has been already discussed, @nand2mario agrees in making a PR adding this feature like an add-on.

Sometime ago I proposed to use an esp32 (with blueretro project) to have bluetooth gamepad support and driving an SPI interface to the FPGA (the SPI controller does not take much resources from the chip)

harbaum, 2024-01-09:
For Blueretro you probably don't even need any changes on FPGA side. It should be possible to make the ESP32 act like a DS2 game pad. But adding an ESP32 sure sounds like a cool idea. The BL616 on the TN20k should also be able to do that. Unfortunately, they left the antenna pin unconnected on the TN20k. I still have to test if the bare pin as an antenna gives at least some weak connectivity. But that's probably futile.

Adding an ESP32 has the disadvantage that Sipeed will probably never consider doing a board with it already integrated. That's why I chose to use the BL616. Sipeed seems to be pretty fond of it ... if only they were more open with the firmware.

Chandler-Kluser, 2024-01-09:
I loved your iniciative in MiSTery Nano and your USB Host implementation for USB keyboards for the project!

About whether Sipeed will like or dislike the idea... Well, that's not actually my concern.

Actually, there is already a repository implementing ESP32 with blueretro for NES in Tang Nano 9k, that's why I have not considered yet putting some additional information in Official Nestang Repository ☺

nand2mario, 2024-01-10:
@harbaum Hi Till, making use of the bl616 makes a lot of sense for SNES as FPGA resource is very tight. I'll find some time to read your bl616-side code. On another note, has Sipeed released the firmware anywhere? Without that I'm kinda worried that this will lead to users with bricked boards. I remember you said in another discussion that they are at least not opposed to the idea.

Thanks a lot.

harbaum, 2024-01-10:
They currently don't allow a source code distribution.

They once asked me to do a binary only library version of the flasher code within my own firmware. Chances are getting worse that this will work as the bouffalo sdk and cherryusb are developed further and the original "20k friend" (the flasher's own USB name), won't build under the latest SDK versions, anymore. 

In order to prevent bricking the flasher a binary only release of the firmware would at least allow you guys to flash the TN20k's BL6161 and have a save way back. This might also work for users who rarely flash their devices. But of course this is no option for us during development.

I am unsure how to proceed from here. On one hand Sipeed wants to develop cool FPGA gaming hardware that can compete with MiST and MiSTer. On the other hand, they are not open enough to allow us to write cores that can competitive.

edanuff, 2024-01-15:
The firmware in the BL616 should be similar to what's in the [RV-Debugger-BL702](https://github.com/sipeed/RV-Debugger-BL702).  It would need to be adapted to the latest [BouffaloSDK SDK ](https://github.com/bouffalolab/bouffalo_sdk).  Perhaps a new project to do this built on top of the [MOs BL616 Example](https://github.com/sipeed/M0s_BL616_example)?  Would be of interest for use with a lot of TN20k projects.

nand2mario, 2024-01-17:
Another approach is to use a softcore CPU in the FPGA. It'll use some LUTs. But the upside is no dependency on Sipeed providing support, and no extra setup work on the user side.

I played with Bruno Levy's FemtoRV32. The quark_bicycle RV32i softcore takes 841 LUTs. Not bad considering it will allow us to do file writes, support exFAT and etc. So I may actually give this a go.

To save resource and provide more flexibility, I'm considering booting the softcore from SD card and use the 2nd SDRAM chip as memory. If successful, it'll give the softcore gigabytes of storage and 32MB of memory. :)

nand2mario, 2024-01-17:
Great. I'll look at your A2FPGA. TN20 has 8MB. I was thinking about Tang Primer 25k. Its memory module has 2*32MB.

nand2mario, 2024-02-01:
I have added a [softcore](https://github.com/nand2mario/snestang/tree/main/src/iosys) to snestang. It currently handles sd card and osd. It boots from the bitstream spi flash. Main memory is in sdram, sharing the same bank with snes cpu. It consumes about 2K LUTs in total, actually smaller than the previous sd+fat32 modules. This looks like the right design before Sipeed opens up the bl616.

Once it is stable, the plan is to use it in nestang too.

fjpolo, 2024-03-19:
@nand2mario what would be the steps to implement the softcore in the NESTang?

nand2mario, 2024-03-20:
The SNESTang [design notes](https://github.com/nand2mario/snestang/blob/main/doc/design.md) and [this article](https://nand2mario.github.io/posts/softcore_for_fpga_gaming/) explains the softcore. 

One issue is that we should either bring the code base more in sync with upstream (NES_MiSTer), which is good for future maitenance. Or we could add the softcore first. Not sure which route is better, or we could do both at the same time.

fjpolo, 2024-03-20:
I guess syncing to NES_MISTer will be the way to go eventually. It's pretty developed and seems to have a huge community around it. Also that's where I got how to integrate VRC6 from Loopy mappers to the NESTang, soo, cheers! On that topic, maybe [this redditor](https://www.reddit.com/r/fpgagaming/comments/18npz1w/comment/ked3x28/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button) could be of help if they started the adventure of migrating the code.

If you ask me what I would do, personally, I'd love to see the softcore implemented. That would make it easier to save/load configurations, savestates, cheats, USB maybe?, and whatever the sofctore can help with. Just having the menu is a ⭐⭐⭐⭐⭐ feature

nand2mario, 2024-03-26:
I agree. I've actually just migrated SNESTang to MiSTer code base. It took about 3 weeks. I'll find some time to do the same thing for NESTang.

nand2mario, 2024-04-07:
The softcore has landed and most of MiSTer NES is ported over.

closed 2024-04-07 by nand2mario