nand2mario/nestang #48: [MMC5 mapper] Faulty emulation of MMC5

fjpolo, 2024-01-13:
I was testing [Celeste Mario Hack](https://www.romhacking.net/hacks/7915/) and found out that apparently MMC5 mapper is not emulated correctly. 

[image]

nand2mario, 2024-01-14:
Reproduced on my setup. MMC5 was not fully tested. I'll take a look.

fjpolo, 2024-03-22:
Note: maybe [this NES_MISTer issue](https://github.com/MiSTer-devel/NES_MiSTer/pull/13/commits/36da4a03e99aa2207673b61a9d94a619d453a430) is the answer. To be checked.

fjpolo, 2024-03-23:
Extra RAM bug fixed but still buggy

[image]
[image]
[image]

nand2mario, 2024-04-07:
This should be fixed in 0.9.

closed 2024-04-07 by nand2mario