nand2mario/nestang #50: Add support for 2048x1536 resolution

siffland, 2024-01-14:
I have one of the ipad retina display's the same use used in Laser Bear's "crt" kit and on aliepxress.  

[Laser Bear](https://www.laserbear.net/products/copy-of-crt-style-9-7-lcd-case)
[aliexpress](https://www.aliexpress.us/item/3256801503431249.html?spm=a2g0o.refund_spm.0.0.5f8872e4Wtxigi&gatewayAdapt=glo2usa)

The issue is the resolution is 2048x1536 (4:3).  This results in the image being squished.  Is there a way to support this resolution, or even a variable and i can compile a version for myself.  In Mister, they created video_mode=13 for this

[Mister video modes](https://mister-devel.github.io/MkDocs_MiSTer/advanced/videomodes/#video-mode-presets-video_mode)

It would be a sweet setup with the nestang (and snestang) and this display.

nand2mario, 2024-01-17:
We can address this once we have a way to save options (#47)

fjpolo, 2024-01-17:
> We can address this once we have a way to save options (#47)

Next topic to work on 😉