nand2mario/nestang #54: Super Mario Bros 2p hack - buggy gameplay

kiwanowski, 2024-03-12:
I was trying to play Super Mario Bros 2p hack - https://www.romhacking.net/hacks/4180/
On the latest version the gameplay is buggy

This romhack is using the VRC6 mapper

fjpolo, 2024-03-15:
Any footage is appreciated, or a better description since `buggy` means something different for every person and every context :) 
Also steps to reproduce please :)

Questions:
- Do earlier versions of the hack work? Or are they also buggy?

I don't think VRC6's integration in the NESTang is 100% working, since I found that loading a VRC6 ROM, going back to the main menu, loading a ROM using an unsupported mapper and then reloading the same VRC6 ROM doesn't work. Sooo there's something there

fjpolo, 2024-03-23:
[NES_MISTer PR#13 commit](https://github.com/MiSTer-devel/NES_MiSTer/pull/13/commits/fd3bb9177954520a763eaa4b3585403c5f8aa905) might be related

fjpolo, 2024-03-23:
@kiwanowski try [this bitstream for TangNano20k](https://github.com/fjpolo/nestang/releases/tag/v20240323)

kiwanowski, 2024-06-03:
Hello, sorry for the late response. Can you send me the file in .fs format?

kiwanowski, 2024-06-22:
@fjpolo