nand2mario/nestang #56: Can't synthesis the project.

radiance28, 2024-03-16:
After the command "gv_sh.exe primer25k.tcl" at the bottom of the console screen: <Error while executing "run all" (file "build.tcl" line 55)>. The same error occurs if you try to synthesize from GowinIDE. How to fix it? I'm using Windows10

fjpolo, 2024-03-19:
A few Qs:

- What branch are you on?
- GoWIN IDE version?
- Can you attach the whole logs? Plus a screenshot of the Errors view

radiance28, 2024-03-19:
1. Branch - Master
2. GoWIN IDE v.1.9.9 with registered license
3. [log_GW_SH.txt](https://github.com/nand2mario/nestang/files/14653624/log_GW_SH.txt)

[image]

When I try to synthesize from GoWIN IDE:

[nes.log](https://github.com/nand2mario/nestang/files/14653689/nes.log)

[image]

fjpolo, 2024-03-19:
`Error EX3937` is telling you that `NESGamepad` module is not found. By some reason NESGamepad.v wasn't added to the project (yours?general?) And that's why it's not compiling

Try adding it from `src/NESGamepad/NESGamepad.v` to the project

radiance28, 2024-03-19:
So what should I do about it?  I just downloaded the source code and wanted to compile it (I didn’t change anything in it) under primet25k

fjpolo, 2024-03-20:
Until [this PR](https://github.com/nand2mario/nestang/pull/57) is merged, `nestang_primer25k` in `master` is broken. You can use this repo and branch: [fjpolo/TangPrimer25k_fixNESGamepad](https://github.com/fjpolo/nestang/tree/fjpolo/TangPrimer25k_fixNESGamepad), but you won't be able to use OEM joysticks.

`NESGamepad` was only developed and tested for `NESTang Nano 20k`, and not implemented nor tested for `NESTang Primer 25k`. The issue you were having is because `Nano20k` and `Primer25k` have different project configurations but share source code. So, `nestang_top.sv` was trying to instantiate `NESGamepad`, a module unknown to `nestang_primer25k` because it wasn't included in `nestang_primer25k.gprj`. The solution would have been to add `NESGamepad.v` to `nestang_primer25.gprj` and all related signals to `nestang.cst`, but since `NESGamepad` for the `Primer25k`wasn't implemented or tested for it, I decided to leave it out of `Primer25k` since I don't have the board to do it.

nand2mario, 2024-03-20:
#57 is merged. Thanks.

radiance28, 2024-03-20:
The project is now being synthesized, thank you!

closed 2024-03-20 by radiance28