nand2mario/nestang #58: [Help wanter][Testers] Multitap Four Score support

fjpolo, 2024-03-20:
⚠ Still in ongoing development ⚠

[fjpolo/TangNano20k_Multitap](https://github.com/fjpolo/nestang/tree/fjpolo/TangNano20k_Multitap) is a branch in development that is supposed to support the [NES Four Score](https://en.wikipedia.org/wiki/NES_Four_Score) by adding two additional OEM controllers.  I got rid of the PS2 and USB controllers, to have more IOs, so it's a OEM-only branch.

I'm looking for testers with more than one joystick (I have only one) in order to test this branch.

What to do:
- Follow the schematics [here](https://github.com/fjpolo/nestang/blob/fjpolo/TangNano20k_Multitap/doc/NESGamepad.md) to connect all gamepad connectors
- Load a Four-Score game, legally obtained, to the SD Card root dir, like [Micro Mages](http://morphcat.de/micromages/)
- Start your NESTang
- Start the game
- Choose 3-4 players
- Press Select+Up to enable Multitap, LED[2] should be ON
- Test it :)

Roet-Ivar, 2024-11-06:
Hi! I'm in the process of setting up my nano 20k for 4-player so I can test this, I'll let you know how it goes.

mariogazziro, 2024-11-06:
wow.. TNMT ahead