nand2mario/nestang #60: [Migration to softcore] Some issues bringing up the uC

fjpolo, 2024-03-28:
@nand2mario, so I've been checking [Design notes for SNESTang](https://github.com/nand2mario/snestang/blob/main/doc/design.md), [Adding a Softcore to SNESTang - part 1](https://nand2mario.github.io/posts/softcore_for_fpga_gaming/) and [Adding a Softcore to SNESTang - part 2](https://nand2mario.github.io/posts/softcore_for_fpga_gaming_part_2/) and trying to adapt `src/iosys` and `firmware` to Nano20k here: [fjpolo/TangNano20k_Softcore](https://github.com/fjpolo/nestang/tree/fjpolo/TangNano20k_Softcore).

So far the idea was to have a simple `while(1)` to blink `LED[0]` every second. For that I just added the `while(1)` before the proper code in `firmware.c`'s `main()`. I also added `#define reg_led            (*(volatile uint32_t*)0x02000060)` in `picorv32.h` and `led_reg = mem_valid && (mem_addr == 32'h 0200_0060);` in `iosys.v`, where I also added `output [1:0] led` and `assign led = led;` to connect it to the board in `nestang_top.sv`.

I tried to add some control logic for the RV RAM, but probably I am missing something there, it might be that iosys is loading the .bin into 0x000000 and that's the NES' RAM. Or it can be that `MemoryController` is incorrect.

It could also be that Flash's not configured/working correctly, since I tried `assign led[0] = flash_loaded;` in `iosys.v` and `led[0]` stays low. I guess the configuration for Flash pins is correct:

[image]

Would you mind taking a quick look at the branch to see if you see something there?

fjpolo, 2024-03-30:
I'll actually try to bring up the softcore + Memory in here: [fjpolo/TangNano20k_softcoreStandalone](https://github.com/fjpolo/nestang/tree/fjpolo/TangNano20k_softcoreStandalone)

fjpolo, 2024-04-02:
Seem that's not synthesizable for the nano:
`ERROR (RP0006) : The number(27548(26308 LUTs, 886 ALUs, 0 ROM16s, 59 SSRAMs)) of logic in the design exceeds the resource limit(20736) of current device`

I thought using a softcore used ~2kLUTS, how is this going from ~10kLUTs to 26308LUTS?

Also, I took snestang's Fw and built it, should it also be flashed @a0x50_0000?

nand2mario, 2024-04-02:
Forgot to mention the current branch is only for primer 25k. I am traveling and will sort out the build issues once I got home in two days.

Regarding the LUT issue. Are you using Gowin IDE >1.9.9? 1.9.9.1 disabled SSRAM for some unknown reason. I am sticking with 1.9.9 for now.

radiance28, 2024-04-02:
[image]

I just checked to compile branch iosys on GoWin1.99 (64bit) (no beta) and everything worked for Primer25k.

[image]

[image]

[image]

nand2mario, 2024-04-02:
Right. The commercial version (1.9.9) works, while the EDU version (1.9.9-beta4) produces the "numbe rof logic exceeded" error.

nand2mario, 2024-04-05:
These should all be fixed as part of v0.9.

The dual-port memory issue was due to an unsupported write mode (read-before-write) for GW2A.

closed 2024-04-05 by nand2mario