nand2mario/nestang #62: Softcore not loading

s3anami, 2024-04-06:
I boot up the device with the new update and it just displays NESTang at the bottom and nothing else
[image]

mariogazziro, 2024-04-06:
I had the same problem, but It Works If you First boot with snestang and then load the nestang only in sdram. I Hope that information helps the developers to figure out what is happening. Probably the nestang is using the same configuration in save directory created by snestang and forgot to be write by nestang.

siffland, 2024-04-06:
I did notice only a snestang.ini gets created.  And both cores (nestang and snestang) pull from the same snestang.ini (if i disable an option in one core, it is disabled in the other (the RAM save option)).

The firmware.bin are the same for both cores (hash i identical).  So in theory to switch the cores you just need to swap out the .fs file (i have been using openFPGAloader on a raspberry pi, which is powering my tang20k, and switched the cores at least 20 times today playing games and have had no issues).

If anyone is interested these are the command I use to load the firmware, they work fine for me (use at own risk, but works fine for me).  Did NOT try on the 25k.

FIRMWARE
`openFPGALoader -b tangnano20k -f firmware.bin -o 0x500000`

SNESTANG
`openFPGALoader -b tangnano20k -f snestang_nano20k.fs`

NESTANG
`openFPGALoader -b tangnano20k -f nestang_nano20k.fs`

nand2mario, 2024-04-07:
I'll look at the loading issue. The firmware is the same, and it does slightly different things depending on the core it is running in. So there could be a dead loop somewhere for nestang that results in the menu not showing up in some cases.

nand2mario, 2024-04-07:
Try this. The problem is that sdram is not properly initialized in some cases. This should fix that.

[nestang_nano20k.zip](https://github.com/nand2mario/nestang/files/14898042/nestang_nano20k.zip)

s3anami, 2024-04-07:
It fixed it for me, thanks.
Does the new OSD support folders now?

siffland, 2024-04-07:
I am using folders, i have them three deep with no issues.

Teku, 2024-04-07:
I was having the same issue where I had to load SNESTang first to get NESTang working. Everything is working now with latest release.

nand2mario, 2024-04-08:
It should be resolved then. Thanks for everyone's help.

closed 2024-04-08 by nand2mario