nand2mario/nestang #64: [0.9][Nano][Cannot load ROM][f_puts failed]  Latest master build won't load ROMs

fjpolo, 2024-04-09:
I checked out latest `master` and built it myself. OSD seems to be working, if I place ROMS inside any `dir`  and press A on that dir, OSD goes back to main menu.

If I press options and try to change any options I get an error:

[image]

If I try to load a ROM from root dir, there's an error:

[image]

FYI I tried using a  8Gb miniSD card formatted as FAT32 in 16kB sectors and also 2048bytes sectors, same result.

Anyone else having similar issues?

fjpolo, 2024-04-09:
Aldo `Error opening director 99` comes up when choosing option `1)`

mariogazziro, 2024-04-10:
Everything Works Fine to me and Also osd in nestang is better than snestang since It keeps transparency

fjpolo, 2024-05-26:
Fixed in [v0.10 - NES/SNES controller suppor](https://github.com/nand2mario/nestang/releases/tag/v0.10)

closed 2024-05-26 by fjpolo