nand2mario/nestang #67: tang nano 20K with blueretro adapter?

wolfshow, 2024-04-20:
any plan to support blueretro for tang nano 20K?

siffland, 2024-04-20:
Also here are the pinouts, in case you are using a different esp32 module compatible with blueretro.  Do not forget power and ground for the esp32 board, i have them tied to the tang 20k (5v and ground).
```
ESP32    TANG 20K
GPIO33 - DS2_CLK
GPIO32 - DS2_CMD
GPIO34 - DS2_ATTN 
GPIO19 - DS2_DATA
```

wolfshow, 2024-04-22:
> Also here are the pinouts, in case you are using a different esp32 module compatible with blueretro. Do not forget power and ground for the esp32 board, i have them tied to the tang 20k (5v and [...]

Thanks! will have a try.