nand2mario/nestang #71: exFAT not work and file list load glitches on FAT32

radiance28, 2024-05-03:
The exFAT file system on NESTang does not work on two of my flash cards; it sees directories and files in the root, but when I try to open it throws me into the main menu. At the same time, SNESTang works without problems. Also, with the FAT32 file system, NESTang displays short file names, like in MS-DOS (8.3), but the same SD card in SNESTang displays files normally. I will demonstrate:

The length of the names is different each time you open the file list:
https://photos.app.goo.gl/769B79r6HN7BLAye6

The same flash card, but with SNESTang:
https://photos.app.goo.gl/ApGmzfNgeffqg2bL7

Because of this, games in NESTang are loaded every once in a while, but most often they are not loaded at all, only ROMs with a minimum size manage to load. Others are stuck at a black screen or frozen loading progress (kB):
https://photos.app.goo.gl/HZNaeEPf8jzeoyS26

[image]

fjpolo, 2024-05-26:
Check if [v0.10 - NES/SNES controller support](https://github.com/nand2mario/nestang/releases/tag/v0.10) fixed it for you, it's the case for me in [#64](https://github.com/nand2mario/nestang/issues/64)

radiance28, 2024-05-27:
> Check if [v0.10 - NES/SNES controller support](https://github.com/nand2mario/nestang/releases/tag/v0.10) fixed it for you, it's the case for me in [#64](https://github.com/nand2mario/nestang/issues/64)

I checked, now everything is readable and loaded into FAT32 and ExFAT. Thank you!

closed 2024-05-27 by radiance28