nand2mario/nestang #73: Compile primer25k error

zzyyll2, 2024-05-20:
[image]

Error occured when i compile it using primer25k type, it seems LUTs over limit then chip.

How to solve this problem?
Thanks.

radiance28, 2024-05-20:
Don't change anything in Project - Configuration, use the default settings. The version of Gowin IDE should preferably be PRO (with free registered license), not EDU

zzyyll2, 2024-05-21:
Got it, I will try it, thanks.

zzyyll2, 2024-05-23:
Not work
1. use default settings.
2. use PRO IDE.

[image]

radiance28, 2024-05-23:
[image]

my version of IDE

nand2mario, 2024-05-23:
@zzyyll2 Use 1.9.9 instead of 1.9.9.03 should solve your problem.

closed 2024-05-23 by nand2mario