nand2mario/nestang #74: What pins do the NES controllers connect to on the Nano?

nikitalita, 2024-05-21:
I see you recently added support for them; which pins do they connect to? I'm going to make my own connector with a breadboard and a voltage logic level converter.

nikitalita, 2024-05-22:
Reading your documentation, I hooked them up to the listed pins, but it doesn't work. I disconnected it and looked at it with a logic analyzer and it seems that those lines are just being pulled high; it should be outputting a latch and clock signal every so often, no?

therealquaid, 2024-05-22:
Are you using the current release?
nand2mario added a few commits since then to get the controller working.
If you compile from source yourself the proper pin out instructions are currently only in the snestang repository:
https://github.com/nand2mario/snestang/blob/main/doc/installation.md

nikitalita, 2024-05-23:
OK, so installing the new core works, and it turns out that hooking up the NES controller directly works; apparently, most of the shift registers in these things could work from anywhere between 3v and 10v.

However, now I have another problem; I am getting constant A and B presses no matter what I do.

Here's why:

[image]

Instead of 8 clock pulses, you're sending 16. Once the 8-bits in the data word have been exhausted, data goes low. Data high means released, low means pressed. 

So, it's counting the rest of the buttons as pressed, meaning that Autofire A and Autofire B are constantly set to pressed. 

However, there's a solution:

[image]

This is an SNES controller schematic. If you'll notice, that last 4 bits are just pulled high, so bits 15:12 on data will always be 1.

Ergo, if we check to see if those bits were pulled high, we'll know if this is an SNES controller. If it isn't, then we consider it an NES controller and discard the extra button states.

closed 2024-05-23 by nand2mario

nand2mario, 2024-05-23:
The fix looks good and is just merged. Thanks @nikitalita.

PR #75 (2024-05-23): Fix NES controllers
Fixes #74