nand2mario/nestang #77: Elite rendering issues

wiebel, 2024-05-24:
This is probably already known, but just in case. Elite's background bitmaps are rendered at an offset of around half a screenheight. I t would be great to be able to play Elite on one of it's original forms.

nand2mario, 2024-05-28:
I'll take a look.

fjpolo, 2024-07-07:
AFAIK Elite is PAL and NESTang is NTSC, so that's why it renders like that. Eg: [Mr Gimmmick PAL running on NTSC](https://misterfpga.org/viewtopic.php?t=4987). 

Idea: Maybe adding a `PAL`/`NTSC` option in `Options` that changes the CPU clock might fix it? PAL runs at ~1,66[MHz) while NTSC runs at ~1,79[MHz]

fjpolo, 2024-07-08:
As I thought, changing `system_type` to `PAL` fixes the issue

[Logic repo:](https://github.com/fjpolo/nestang/tree/fjpolo/PAL_NTSC)
[Firmware repo](https://github.com/fjpolo/snestang/tree/fjpolo/PAL_NTSC)
[Nano20k  release](https://github.com/fjpolo/nestang/releases/tag/v20240708)