nand2mario/nestang #84: [Rewind][WIP]

fjpolo, 2024-06-16:
For my own understanding and enjoyment, I'm working on a Rewind capability for the NESTang, still WIP, who knows if ti will even work..

[fjpolo/TangNano20k_rewind_v](https://github.com/fjpolo/nestang/tree/fjpolo/TangNano20k_rewind_v)

fjpolo, 2024-07-02:
Update: Still WIP since it's a bit more work than I though, so still on it 🐱‍💻

siffland, 2024-08-14:
Can fast forward be added to this.  Or should it be a separate Issue (making dialog and pre-rendered scenes go faster, useful on slow dialog).

fjpolo, 2024-09-04:
@siffland different topic in my opinion. Rw saves all register states as backup to load them later if needed. I guess Fw would work changing system clock to x2, x4, ..

djjohnson13p, 2024-09-10:
@fjpolo any progress on this front?

fjpolo, 2024-09-20:
@djjohnson13p not working on it currently :)