nand2mario/nestang #90: newbie question about nestang_nano20k.gprj

oldoldman, 2024-09-28:
Hi guys,
When run synthsizing of the nestang_nano20k.gprj, I got the following error
```
ERROR (RP0006) : The number(27765(26541 LUTs, 882 ALUs, 0 ROM16s, 57 SSRAMs)) of logic in the design exceeds the resource limit(20736) of current device
```
How does this happen? Many thanks!

fjpolo, 2024-10-01:
A couple followup questions :)

1. What branch are you on?
2. What version of GOWIN IDE are you using? EDU? Licensed?

oldoldman, 2024-10-03:
I'm using nestang_nano20k.gprj on the master branch. And the IDE is EDU. BTW, I've tried nestang-0.7.2rc , which works fine.

nand2mario, 2024-10-04:
Make sure you use Gowin 1.9.9. I use the Pro version (the Edu version should work). 1.9.10 does not work as Gowin somehow disabled SSRAM support in this release for some reason, which results in excessive resource usage.

closed 2024-10-19 by oldoldman