nand2mario/nestang #93: "Cannot load rom" message

bepconcept, 2024-10-16:
I have a Tangnano 20k, with 0.11 version, nestang_nano20.fs and firmware flashed.
When I try to load the NES file from the MMC card, I get a "Cannot load rom" message.
What could be the reason ?
It only worked once, I and was able to play. But only once :(
I tried to format the MMC card on Linux and on Windows : same result.
Would it be possible to use the onboard leds to display an error code ?

fjpolo, 2024-10-19:
I had to try different microSD cards until I found one that worked, give another one a try just in case

bepconcept, 2024-10-19:
@fjpolo, thanks for the tip : I changed for a new 32Gb card and it works :-)
The loading of binary files in the /roms directory was working, so I discarded to try another card ;-)

closed 2024-10-19 by bepconcept