nand2mario/nestang #101: Contra (Japan version) has no screen after loading...

szuping, 2024-12-14:
Contra (Japan version) has no screen after loading...

Contra (US version) can be executed normally

NESTang v0.12
Tang Primer 25K

roundy07, 2024-12-16:
I suspect it’s a mapper issue. 

Akumajo Densetsu does the same, the Japanese version uses a custom Konami VRC6 chip whereas Castlevania 3 in USA and Europe use MMC5

I know Japanese Contra uses a custom chip whereas the US version uses standard Nintendo mappers (US is mapper 2, JAP is Mapper 23 with the VRC II chip)

szuping, 2024-12-17:
> I suspect it’s a mapper issue.
>
> Akumajo Densetsu does the same, the Japanese version uses a custom Konami VRC6 chip whereas Castlevania 3 in USA and Europe use MMC5
>
> I know Japanese Contra uses a [...]

Perhaps as you described, I found that many "Japanese versions" of NES games cannot be executed, such as:
Ganbare Goemon! - Karakuri Douchuu
Ganbare Goemon 2
Salamander....

There are also many games with abnormal graphics...

roundy07, 2024-12-17:
> Perhaps as you described, I found that many "Japanese versions" of NES games cannot be executed, such as: Ganbare Goemon! - Karakuri Douchuu Ganbare Goemon 2 Salamander....
>
> There are also many [...]

All of those Japanese games you mention are Konami and use custom chips not present in US/EU releases

NES Cart DB is a great resource for looking up cart info
https://nescartdb.com

It would be useful if there was a list of implemented mappers on NESTang, the MiSTer NES core has this documented on their GutHub so something similar here would be ideal

fjpolo, 2024-12-23:
[These](https://github.com/nand2mario/nestang/tree/master/src/mappers) are the supported mappers rn. If anyone wants to test them and make a list, please add it to a new file MAPPERS.md in roopt folder in a PR

szuping, 2024-12-23:
> [這些](https://github.com/nand2mario/nestang/tree/master/src/mappers)是支援的映射器。

In addition, I encountered an exception in exfat. The file list could not be displayed correctly. There were missing files or files could not be displayed. I replaced 5-6 TD cards, but it was not a TF card problem. As long as the file list was formatted to FAT32, the file list could be displayed normally. .